locusrobotics/robot_navigation #2: configuration issue

henoSH, 2018-01-18:
Hi,

when i m trying to use this local planner, i faced some problems. I m getting the following error whem i started the move base:
Failed to create the nav_core_adapter/LocalPlannerAdapter planner, are you sure it is properly registered and that the containing library is built? Exception: According to the loaded plugin descriptions the class nav_core_adapter/LocalPlannerAdapter with base class type nav_core::BaseLocalPlanner does not exist. Declared types are  base_local_planner/TrajectoryPlannerROS dwa_local_planner/DWAPlannerROS nav_core_adapter::LocalPlannerAdapter

i configured as following the local planner im move base:
base_local_planner: "nav_core_adapter/LocalPlannerAdapter"
LocalPlannerAdapter:
  planner_name: "dwb_local_planner/DWBLocalPlanner"

DLu, 2018-02-01:
Many old `move_base` configurations use a [deprecated format](http://wiki.ros.org/pluginlib#pluginlib.2BAC8-pluginlib_groovy.Legacy_.22Lookup_Name.22) for specifying plugins.

As specified in this [doc](https://github.com/locusrobotics/robot_navigation/tree/master/nav_core_adapter), if you specify your planner names using two colons instead of slashes, you should have better results.

henoSH, 2018-02-13:
Thank you i did and it works fine I mean i could launch the move base but when i send a goal move base crashed

ngocnt2607, 2020-11-28:
hi I have a same error, Can you help me to fix it. Thank you!
[FATAL] [1606559253.316794159]: Failed to create the nav_core_adapter::LocalPlannerAdapter planner, are you sure it is properly registered and that the containing library is built? Exception: According to the loaded plugin descriptions the class nav_core_adapter::LocalPlannerAdapter with base class type nav_core::BaseLocalPlanner does not exist. Declared types are  base_local_planner/TrajectoryPlannerROS dwa_local_planner/DWAPlannerROS teb_local_planner/TebLocalPlannerROS
[move_base_node-1] process has died [pid 10324, exit code 1, cmd /home/justin/catkin_ws/devel/lib/move_base/move_base map:=map odom:=/odom_comb marker:=move_base_node/DWBLocalPlanner/markers __name:=move_base_node __log:=/home/justin/.ros/log/c5d0f590-3160-11eb-be62-6036dd83cd72/move_base_node-1.log].
log file: /home/justin/.ros/log/c5d0f590-3160-11eb-be62-6036dd83cd72/move_base_node-1*.log

DLu, 2020-12-01:
@ngocnt2607 You have a different error, because LocalPlannerAdapter is not on the list of declared types. Please open a new ticket if you are still have the issue, and be sure to include how you installed robot_navigation and your OS/ROS versions.

closed 2020-12-01 by DLu