locusrobotics/robot_navigation #5: Minimum example and parameter list?

SteveMacenski, 2018-02-13:
Hi,

I was wondering if there was a ROS wiki styled page in the works for the adapters and DWB planner configuration. I would the brief explanation on using the adapter in the yaml format but I could not find an exhaustive list of the parameters available, some description, and default information. 

I'm particularly interested in documentation so I can tune DWB in similar ways you've presented in your ROSCon talk with precise and "drunken" path following

SteveMacenski, 2018-07-29:
Pinging back on this. I saw there's more to come in the upcoming ROSCon talk and it would be good if this had some better docs so that its approachable to the majority of users

BriceRenaudeau, 2019-07-17:
Hello, 
I'm currently using move_base and like you I find it lacking in many ways.
I would like to try your robot_navigation but I don't know where to start.
It would be great to add some example in the docs (launch, config, params, ...)

Combinacijus, 2021-03-09:
Only parameter list I could find is [here](https://answers.ros.org/question/341273/no-obstacle-avoidance-with-dwb-controller/) but still can't find explanation what those parameters do although many parameters are from [DWA planner](http://wiki.ros.org/dwa_local_planner)