locusrobotics/robot_navigation #10: Local plan with 0 poses.

veeraragav, 2018-09-18:
I followed the procedure in the documentation to use this planner with nav_core. The plugin was properly loaded. I get this error **"computeVelocityCommands exception: Resulting plan has 0 poses in it."**, when i give a goal.

crdelsey, 2019-01-09:
This can happen if DWB thinks the robot pose is too far from the plan generated by the global planner. You can try setting the `prune_plan` parameter to false to work around that.

Alternatively, if the costmap doesn't show free space around the path and current robot pose, you can get this issue as well, if I'm not mistaken.

veeraragav, 2019-01-23:
The robot pose is with respect to map frame and the global plan is with respect to the odom frame. My robot's initial position is away from the map frame whic i set using the rviz. When i set the initial position of the robot in rviz, the robot's base link along with the odom frame moves there. Now i set the goal position. In this case the robot's pose is with respect to the map frame and the global plan coordinates are with respect to odom frame. In this case the distance calculated by this function `getSquareDistance()` becomes greater than the default prune distance which is 1m. How can i make it work without disabling the pruneplan?

veeraragav, 2019-01-23:
I use this with nav_core using the adapter. The global_frame for local costmap is set to /odom. The possible solution could be switching it to /map. I had used obsctacle and inflation layer in the local costmap. Will changing it from /odom to /map cause any problem with the working?