locusrobotics/robot_navigation #22: nav core adapter package build fails because of costmap2Dros

sachininder10, 2018-12-03:
/home/inderjeet/Projects/ros_ws/src/robot_navigation/nav_core_adapter/src/costmap_adapter.cpp: In member function ‘virtual void nav_core_adapter::CostmapAdapter::initialize(const ros::NodeHandle&, const string&, TFListenerPtr)’:
/home/inderjeet/Projects/ros_ws/src/robot_navigation/nav_core_adapter/src/costmap_adapter.cpp:76:81: error: no matching function for call to ‘costmap_2d::Costmap2DROS::Costmap2DROS(const string&, std::__shared_ptr_access<tf::TransformListener, (__gnu_cxx::_Lock_policy)2, false, false>::element_type&)’
   costmap_2d::Costmap2DROS* costmap_ros = new costmap_2d::Costmap2DROS(name, *tf);
                                                                                 ^
In file included from /home/inderjeet/Projects/ros_ws/src/robot_navigation/nav_core_adapter/include/nav_core_adapter/costmap_adapter.h:40:0,
                 from /home/inderjeet/Projects/ros_ws/src/robot_navigation/nav_core_adapter/src/costmap_adapter.cpp:35:
/home/inderjeet/Projects/ros_ws/src/robot_navigation/costmap_2d/include/costmap_2d/costmap_2d_ros.h:82:3: note: candidate: costmap_2d::Costmap2DROS::Costmap2DROS(const string&, tf2_ros::Buffer&)
   Costmap2DROS(const std::string &name, tf2_ros::Buffer& tf);
   ^~~~~~~~~~~~
/home/inderjeet/Projects/ros_ws/src/robot_navigation/costmap_2d/include/costmap_2d/costmap_2d_ros.h:82:3: note:   no known conversion for argument 2 from ‘std::__shared_ptr_access<tf::TransformListener, (__gnu_cxx::_Lock_policy)2, false, false>::element_type {aka tf::TransformListener}’ to ‘tf2_ros::Buffer&’
make[2]: *** [CMakeFiles/costmap_adapter.dir/src/costmap_adapter.cpp.o] Error 1
make[1]: *** [CMakeFiles/costmap_adapter.dir/all] Error 2

DLu, 2018-12-03:
If compiling with melodic, please use the [tf2 branch](https://github.com/locusrobotics/robot_navigation/tree/tf2)

sachininder10, 2018-12-07:
Hey David,
                 I tried compiling with the tf2 branch but now the locomotor package gives the same error.
Errors     << locomotor:make /home/inderjeet/Projects/ros_ws/logs/locomotor/build.make.000.log
/home/inderjeet/Projects/ros_ws/src/robot_navigation/locomotor/src/locomotor.cpp: In constructor ‘locomotor::Locomotor::Locomotor(const ros::NodeHandle&)’:
/home/inderjeet/Projects/ros_ws/src/robot_navigation/locomotor/src/locomotor.cpp:58:30: error: ‘TransformListener’ is not a member of ‘tf’
   tf_ = std::make_shared<tf::TransformListener>(ros::Duration(10));
                              ^~~~~~~~~~~~~~~~~
/home/inderjeet/Projects/ros_ws/src/robot_navigation/locomotor/src/locomotor.cpp:58:30: note: suggested alternative: ‘TransformLists’
   tf_ = std::make_shared<tf::TransformListener>(ros::Duration(10));
                              ^~~~~~~~~~~~~~~~~
                              TransformLists
/home/inderjeet/Projects/ros_ws/src/robot_navigation/locomotor/src/locomotor.cpp:58:30: error: ‘TransformListener’ is not a member of ‘tf’
/home/inderjeet/Projects/ros_ws/src/robot_navigation/locomotor/src/locomotor.cpp:58:30: note: suggested alternative: ‘TransformLists’
   tf_ = std::make_shared<tf::TransformListener>(ros::Duration(10));

DLu, 2018-12-07:
Somehow I missed that in the initial translation. The good news is that I already fixed it here: https://github.com/locusrobotics/robot_navigation/commit/3115795eea94a78cae3165cc2533593770666243

Please pull the latest version of the code.

sheikspeare, 2019-09-11:
Hi DLu, I'm still having the same issue ...

DLu, 2019-09-11:
Hi @sheikspeare - Please post your rosdistro and what branch of the code is checked out on your machine.