locusrobotics/robot_navigation #26: path_distance_bias parameter is ignored by the PathDist critic

mbharatheesha, 2019-01-18:
The `path_distance_bias` parameter is not used in the `PathDist` critic function, although it is moved to a new parameter within the PathDist critic namespace. Is this intentional?

DLu, 2019-01-30:
So DWA has the [path_distance_bias](https://github.com/ros-planning/navigation/blob/6ed1eb3ad5061fa260342e91678b9c7b7d8c09f7/dwa_local_planner/src/dwa_planner.cpp#L65) parameter that is used to control how much the path distance is weighted. 

DWB uses a more generic `scale` parameter for each critic. If you're loading a DWA configuration with DWB the parameter [is moved to maintain the functionality](https://github.com/locusrobotics/robot_navigation/blob/b42ec7f26e4619f83e3e9e9fd729699d52f1410c/dwb_local_planner/src/backwards_compatibility.cpp#L73)

So unless I misunderstand you, it is intentional. The parameter is used but not under the same name.

mbharatheesha, 2019-02-20:
Thanks for your response. I verified the parameter usage with the `scale` name and it works as intended.

closed 2019-02-20 by mbharatheesha