locusrobotics/robot_navigation #42: [Question] I want to add a hook for switching local planner

ompugao, 2019-05-09:
thank you for your great package.

Just a question.
I know that pluginmux has an ability to add a callback before switching plugin.
But there is no api for the user of 'locomotor' library to add some hooks,
because setSwitchCallback is called inside constructor and local_planner_mux_ is protected.
https://github.com/locusrobotics/robot_navigation/blob/master/locomotor/src/locomotor.cpp#L65

Is it correct to create a new class which inherets 'Locomotor' to achieve that?

ompugao, 2019-05-09:
If yes, I'm happy if this member variable is std::shared_ptr<Locomotor> and I can assign a customized Locomotor instance to it.
https://github.com/locusrobotics/robot_navigation/blob/master/locomotor/src/double_thread_locomotor.cpp#L195

DLu, 2020-07-02:
So generally, yes, I think the approach would be to extend the class. 

However, is the goal to add hooks for before the planner is switched via the ROS service switching? Because otherwise I would just modify the behavior when `useLocalPlanner` is called.