locusrobotics/robot_navigation #45: Porting dlux_global_planner to ROS 2 Dashing

sloretz, 2019-05-22:
I'm looking into what it would take to port `dlux_global_planner` to ROS 2 Dashing. Hopefully you don't mind if I ask a few questions.

* Is someone already working on this?
  * If yes and they stopped, is their progress available to start from?
* What needs to be ported before `dlux_global_planner`?
  * It looks like: `nav_core2`, `nav_grid`, `dlux_plugins`, `global_planner_tests`, `nav_grid_iterators`, `nav_grid_pub_sub`
    * Are any of these already ported?
* How much of `nav_core2` overlaps with `nav2_tasks`?
  * Should a ROS 2 port of `dlux_global_planner` use `nav_core2::GlobalPlanner` or `nav2_tasks::ComputePathToPose...`?
* Where should ports of these packages live?
  * `ros2` branch on this repo?
  * ros-planning/navigation2?
* Anything else I should know or questions I should have asked?

orduno, 2019-06-19:
> Should a ROS 2 port of dlux_global_planner use nav_core2::GlobalPlanner or nav2_tasks::ComputePathToPose...?

Nav2 interfaces with planning and control through ROS2 actions. Dlux would need an action service for `nav2_msgs::action::ComputePathToPose`.

Just a few random things to consider:
* Lifecycle enabled. I would also consider what design changes might be needed (if any at all) to make it LIfecycle friendly, and start with that in mind during porting process.
* World representation. Another thing to consider, in Nav2 we are trying to separate the [world representation](https://github.com/ros-planning/navigation2/issues/565) from the planners and controllers. Currently, `nav2::navfn_planner` gets the costmap from a service call to world model.
* Costmaps. We ported @DLu costmap package about a year ago, and I don't think we have kept up with updates and probably diverged a bit.

crdelsey, 2019-06-19:
> Where should ports of these packages live?

@sloretz I'll leave that up to you and @DLu. If you decide you want to put this in navigation2, we'd be happy to add it and will support you as much as possible. My desire is to integrate this and navigation2 as much as possible, and where not possible, make it easy to swap back and forth between them.