locusrobotics/robot_navigation #46: Installing dwb_local_planner with apt install

JulianS-Uni, 2019-05-28:
I am using ros kinetic on an Ubuntu 16.04 64bit. I tried to install dwb_local_planner and nav_core_adapter with sudo apt install ros-kinetic-... which installs both packages cleanly. But then when I tried to use them in my usual navigation stack (the move_base one) I got an error that both package are not initialised as plugins to the nav-core which I could confirm after checking with rospack plugins --attrib=plugin nav_core. Shouldn't those packages register themselves automatically to nav-core as plugins? 

I workaround this by downloading the current master (I think its version 2.5) and build them with catkin_make which worked after I deleted all the locomo packages which gave me compilation errors mainly because virtualenv or anything was missing (I think that's another issue).

SteveMacenski, 2019-05-28:
it is not a `nav_core` plugin. https://github.com/locusrobotics/robot_navigation/blob/master/dwb_local_planner/plugins.xml#L3 use the shim https://github.com/locusrobotics/robot_navigation/tree/master/nav_core_adapter

DLu, 2019-05-28:
@SteveMacenski is right. Use the documentation in the second link to load the planner into `move_base.

closed 2019-05-28 by DLu

JulianS-Uni, 2019-05-28:
> it is not a nav_core plugin. 

Yes you are right `dwb_local_planner` is not a `nav_core` plugin but `nav_core_adapter` should be, right?
As i mentioned i got it to work by getting the latest versions from [here](https://github.com/locusrobotics/robot_navigation/releases/tag/0.2.5), i am just curious why the version of `nav_core_adapter` one can install with `apt install` is not recognized as a `nav_core` plugin which i checked with `rospack plugins --attrib=plugin ` whereas `nav_core_adapter` i got from github is.

DLu, 2019-06-07:
The version of `nav_core_adapter` in the binaries is [0.2.5](https://github.com/ros/rosdistro/blob/master/kinetic/distribution.yaml#L10799) as well, so I am confused. My current theory is that maybe if you had done `rospack profile` then `rospack plugins --attrib=plugin nav_core` it might have worked, but can't be sure.