locusrobotics/robot_navigation #48: what are the advantages of DWB planner?

bxc237, 2019-08-19:
Hi,
I would like to know that comparing to the original DWA planner, what makes the DWB planner be a better option for the navigation stack? Thanks!

airuchen, 2019-10-16:
Hi there, 
I found this comparison quite useful. You may want to give it a look.
[https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/README.md](url)

ninamwa, 2019-12-06:
Hi. I am confused about this sentence from the link: DWB was meant to replace both the planners in the navigation repository and provides an implementation of both DWA and Trajectory Rollout.

Is DWB an improved version of DWA or is it a combination of DWA and Trajectory Rollout?

DLu, 2020-07-02:
For the fullest comments, please check out [my talk here](https://vimeo.com/236487972)

In essence, DWB was written to replace and be backward compatible with `dwa_local_planner`, and since that package implemented algorithms equivalent to both DWA and trajectory rollout (I believe), DWB can do both as well.

closed 2020-07-02 by DLu