locusrobotics/robot_navigation #62: DluxGlobalPlanner: path too close to obstacle

dschnabel, 2020-08-06:
I'm using the DluxGlobalPlanner with AStar and GradientPath to calculate global paths. The problem is that the resulting path is too close to obstacles:
[image]

I was playing around with `neutral_cost` and `scale` but the few values I tried didn't make a huge difference on the path. What parameters should be looked at to get a path closer to the one I want?

Used parameters:
```
DluxGlobalPlanner:
  neutral_cost: 50
  scale: 3.0
  unknown_interpretation: lethal
  path_caching: false
  improvement_threshold: -1.0
  publish_potential: false
  print_statistics: false
  potential_calculator: dlux_plugins::AStar
  traceback: dlux_plugins::GradientPath

  # AStar
  manhattan_heuristic: false
  use_kernel: true
  minimum_requeue_change: 1.0

  # GradientPath
  step_size: 0.5
  lethal_cost: 250.0
  iteration_factor: 4.0
  grid_step_near_high: false
```

dschnabel, 2020-08-08:
Ok, got it the way I wanted by increasing the value for `cost_scaling_factor` in the `inflation_layer` of my `global_costmap`.
Old values:
```
  inflation_layer:
    cost_scaling_factor: 2.20
    inflation_radius: 1.75
```
New values:
```
  inflation_layer:
    cost_scaling_factor: 10.0
    inflation_radius: 2.0
```
This makes the DluxGlobalPlanner calculate a path which is far enough away from obstacles.

closed 2020-08-08 by dschnabel

tugbakara, 2023-07-05:
Is there any solution in source code? Because I already tried all params to achieve yours but I failed again and again . I tried GIMP to draw map for dangereous corners but when someone stands on the robot and robot try to create and follow new path it gets stack in location that I draw in GIMP.