locusrobotics/robot_navigation #67: [Question, Enhancement] - limit centripetal acceleration during turns

Thazz, 2020-10-19:
It would be nice to have an option to limit robot's velocity during turns, as currently robot doesn't decrease its `vel_x` value when turning. My question is where it would be better to apply these limits (e.g `acc_lim_cent`):
- Write my own Critic that scores trajectories based on centripetal acceleration.
- Modify `computeNewVelocity()` function in trajectory generator (as this is where `x`, `y` and `theta` acceleration limits are currently applied).

DLu, 2020-10-23:
Hey @Thazz,
I agree with what you're saying. There's basically two ways to do it. 
 * If you have a robot that you absolutely do not want to above a certain forward velocity while also turning, that would probably be best handled by a new `TrajectoryGenerator` plugin. 
 * You could also make it in its own critic that could either score or reject those velocities. 

I'd probably go with the latter since it seems the most flexible, but either could work. 

Does that help?

Thazz, 2020-10-26:
Thanks @DLu 

I've chosen the "new `TrajectoryGenerator`" approach and it does it's job well.

closed 2020-10-26 by Thazz

Nurullahdmrly, 2021-06-29:
Hey @Thazz 
Have you added a constraint parameter for the returns or have you come up with a solution for it(acc_lim_cent) ? I want to do the same and if you have solution can you share it with me?