locusrobotics/robot_navigation #82: Dynamic changing map with GlobalPlannerAdapter

lantica, 2021-01-05:
Hi, I am using the ordinary movebase with the Dluxglobalplanner and DWBlocalplanner with the adapter.
I was trying to update the costmap dynamically in the runtime by publishing new map topic.
Although the costmap was updated,  it seems the planner didn't update and keeps return error when I try to set new goal.
Any ideas?

P.S. I used to work with locomotor before, however segfault will occur if the distance between start pose and goal pose is too far. Anyone also have this issue before?

bren-dusty, 2021-01-20:
I had the same problem and found a work around.  Maybe david has a proper solution.

In the costmap adapter.cpp it copies the map in line 71.  Thus the controller will not dynamically update with updates to the controller. So to solve this I added thes to lines:

in costmap_adapter.com line 91 (update())  -> costmap_ = costmap_ros_->getCostmap();

In dlux_global_planner.cpp line 106 (makeplan()) ->   costmap_->update();

bren-dusty, 2021-01-20:
I had the same problem and found a work around. Maybe david has a proper solution.

In the costmap adapter.cpp it copies the map in line 71. Thus the controller will not dynamically update with updates to the costmap. So to solve this I added these to lines:

in costmap_adapter.com line 91 (update()) -> costmap_ = costmap_ros_->getCostmap();

In dlux_global_planner.cpp line 106 (makeplan()) -> costmap_->update();

lantica, 2021-01-21:
@bren-dusty It works, thank you!
Previously I was using a stupid work around that extend every map into same size.
Would you consider to make a PR?

lantica, 2021-01-21:
One interest thing, after new costmap is published, the global planner will provide a wrong path in the beginning. Like the picture below. The global path is just a straight line between start and end. 
[image]

It can calculate the correct path after 10 -15 second.
[image]

bren-dusty, 2021-01-21:
> @bren-dusty It works, thank you!
> Previously I was using a stupid work around that extend every map into same size.
> Would you consider to make a PR?

I think this is more a hack then a real solution.  But I'm still working on the code, so if I find a more elegant solution I will make a PR.