locusrobotics/robot_navigation #86: [robot_nav_rviz_plugins] Polygon2D crashed under Noetic

Timple, 2021-03-29:
Terminal output:

```
qt5ct: using qt5ct plugin
[ INFO] [1617030354.240585298]: rviz version 1.14.6
[ INFO] [1617030354.240622348]: compiled against Qt version 5.12.8
[ INFO] [1617030354.240630826]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617030354.249384809]: Forcing OpenGl version 0.
[ INFO] [1617030354.654440123]: Stereo is NOT SUPPORTED
[ INFO] [1617030354.654605205]: OpenGL device: Quadro M1000M/PCIe/SSE2
[ INFO] [1617030354.654636445]: OpenGl version: 4.6 (GLSL 4.6).
qt5ct: D-Bus global menu: no
[ WARN] [1617030358.204115709]: OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
Aborted (core dumped)
```

To reproduce: Open rviz and add the Polygon2D visualization.

Any more things I can do to help debugging?

Timple, 2021-03-29:
I used `backward_ros` to create a stacktrace when building the latest master:

```
qt5ct: using qt5ct plugin
[ INFO] [1617030578.982143846]: rviz version 1.14.6
[ INFO] [1617030578.982186041]: compiled against Qt version 5.12.8
[ INFO] [1617030578.982198356]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617030578.991089453]: Forcing OpenGl version 0.
[ INFO] [1617030579.425028557]: Stereo is NOT SUPPORTED
[ INFO] [1617030579.425180197]: OpenGL device: Quadro M1000M/PCIe/SSE2
[ INFO] [1617030579.425243426]: OpenGl version: 4.6 (GLSL 4.6).
qt5ct: D-Bus global menu: no
[ WARN] [1617030583.940529549]: OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f70a138da65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f70a13d02b5, in QWidget::event(QEvent*)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f70a1484034, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f70a1483e72, in 
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f70a1482a2d, in 
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f70a1482805, in QAbstractButton::clicked(bool)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7f70a0f8c2ff, in QMetaObject::activate(QObject*, int, int, void**)
#24   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1b94005, in rviz::DisplaysPanel::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)
#23   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bcec97, in rviz::DisplaysPanel::onNewDisplay()
#22   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1ced8b6, in rviz::VisualizationManager::createDisplay(QString const&, QString const&, bool)
#21   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bcb220, in rviz::DisplayGroup::createDisplay(QString const&)
#20   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bc97ac, in rviz::ClassIdRecordingFactory<rviz::Display>::make(QString const&, QString*)
#19   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1babe70, in rviz::DisplayFactory::makeRaw(QString const&, QString*)
#18   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bafb1e, in rviz::PluginlibFactory<rviz::Display>::makeRaw(QString const&, QString*)
#17   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bb27d8, in pluginlib::ClassLoader<rviz::Display>::createUnmanagedInstance(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#16   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bb59b9, in rviz::Display* class_loader::MultiLibraryClassLoader::createUnmanagedInstance<rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#15   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bb9c33, in rviz::Display* class_loader::ClassLoader::createUnmanagedInstance<rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#14   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bbbfa8, in rviz::Display* class_loader::ClassLoader::createRawInstance<rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
#13   Object "/home/tim/ros/harvey/system/devel/.private/rviz/lib/librviz.so", at 0x7f70a1bbe227, in rviz::Display* class_loader::impl::createInstance<rviz::Display>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, class_loader::ClassLoader*)
#12   Object "/opt/ros/noetic/lib/librobot_nav_rviz_plugins.so", at 0x7f706d2789bf, in class_loader::impl::MetaObject<robot_nav_rviz_plugins::PolygonDisplay, rviz::Display>::create() const
#11   Object "/opt/ros/noetic/lib/librobot_nav_rviz_plugins.so", at 0x7f706d2698d3, in robot_nav_rviz_plugins::PolygonDisplay::PolygonDisplay()
#10   Object "/opt/ros/noetic/lib/librobot_nav_rviz_plugins.so", at 0x7f706d29bf4d, in robot_nav_rviz_plugins::PolygonMaterial::PolygonMaterial()
#9    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0", at 0x7f709ff9d281, in Ogre::MaterialManager::create(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, Ogre::ManualResourceLoader*, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<c [... cut ...]
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f70a0ba14e6, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f70a0ba147b, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f70a0b95950, in 
#1    Source "/build/glibc-eX1tMB/glibc-2.31/stdlib/abort.c", line 79, in abort [0x7f70a090f858]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0x7f70a093018b]
Aborted (Signal sent by tkill() 468936 1000)
Aborted (core dumped)
```

DLu, 2021-04-17:
Using 20.04 I assume?

DLu, 2021-04-17:
Also, does it crash when you don't supply data, or only when you do? If the latter, please send me a bag with the polygon data.

Timple, 2021-04-26:
It crashes without supplying any data. Just a roscore and rviz with the plugin enabled.

DLu, 2021-04-29:
Okay, I've replicated on my machine, both from binary and source. I'm unclear why this problem did not surface during the initial testing. Ogre is black magic to me. 

The annoying thing is I've been here before, and I feel there was some reason why [on this line](https://github.com/locusrobotics/robot_navigation/blob/ea31b8ac08b25c37015b794108c2121b5d84cce9/robot_nav_tools/robot_nav_rviz_plugins/src/polygon_parts.cpp#L162) I used `"rviz"` instead of `"robot_nav_rviz_plugins"` or `Ogre::ResourceGroupManager::DEFAULT_RESOURCE_GROUP_NAME`. With the latter, it doesn't crash immediately. Will investigate further.

Timple, 2021-04-29:
Awesome, thanks for looking into it!

Timple, 2021-06-03:
Seems related: https://github.com/jsk-ros-pkg/jsk_visualization/issues/799 also polygons involved.

KenYN, 2021-07-21:
> The annoying thing is I've been here before, and I feel there was some reason why on this line I used "rviz" instead of "robot_nav_rviz_plugins" or Ogre::ResourceGroupManager::DEFAULT_RESOURCE_GROUP_NAME. With the latter, it doesn't crash immediately. Will investigate further.

This change works for me to fix https://github.com/jsk-ros-pkg/jsk_visualization/issues/799.

chrisppp4, 2021-08-06:
> > The annoying thing is I've been here before, and I feel there was some reason why on this line I used "rviz" instead of "robot_nav_rviz_plugins" or Ogre::ResourceGroupManager::DEFAULT_RESOURCE_GROUP_NAME. With the latter, it doesn't crash immediately. Will investigate further.
> 
> This change works for me to fix [jsk-ros-pkg/jsk_visualization#799](https://github.com/jsk-ros-pkg/jsk_visualization/issues/799).

Could you please provide some detailed information about fixing jsk_rviz_plugins crash problem?
I'll be really appreciated cuz this problem is really annoying.
PolygonArray under jsk_rviz_plugins works just fine in ros-melodic.
But it just doesn't work in ros-noetic.

PR #92 (2022-04-11): Fix Polygon2D crash under noetic
I've basically applied the same fix here as https://github.com/jsk-ros-pkg/jsk_visualization/commit/f8e7a3f00d25d4af4c12220a8c94293004a7adf4

Fixes #86
closed 2022-04-11 by DLu